senseshift/senseshift-firmware #4: Install guide

Hotohori, 2022-09-30:
We really need one, I only found how I need to build it out from analyzing the files and trial and error (I guess because I use other software versions). But I get it to build the code.

What you need (for others):
1. Vistual Studio Code
2. Inside VSC the Extensions PlatformIO IDE, Espressif IDF and maybe Arduino(?), at the last not sure, I simply installed it.
3. Clone the Git Code from here to your drive and open it with PlatformIO, on their website is a tutorial how.

I get an error "src_filter is missing inside [Common] in platformio.ini" (something like that). I copied build_src_filter inserted it before and remove "build_", it worked.

But when I want to upload it, it load all files from all bHaptics product one after another up, so you need to make a backup of ini/bhaptics.ini and delete all except the one you want to build and upload.

And after this... holly cow, it works perfect in the bHaptics Feedback Test! Couldn't find an issue here. Now I can start to build my vest in the next weeks. I tested it with the Tactosi for feets, because it only has 3 rumble motors, easier to test. All 3 ports worked as they should inclusive the intensity.

leon0399, 2022-10-03:
I've added some kind of very rudimental documentation, it's better than nothing

sanjay900, 2022-10-12:
>I get an error "src_filter is missing inside [Common] in platformio.ini" (something like that). I copied build_src_filter inserted it before and remove "build_", it worked.

Seems like you might just have an old version of platformio? The newer ones removed `src_filter` and use `build_src_filter` instead, and complain if you use `src_filter`

leon0399, 2022-10-17:
I suppose this exact error happened because `common` configuration really does not have `src_filter`, it is only used now to store basic options

sanjay900, 2022-11-02:
This problem should be resolved now :)

leon0399, 2022-11-23:
BTW, now there is this guide: https://openhaptics.github.io/docs/flashing/platformio

closed 2022-12-12 by leon0399