senseshift/senseshift-firmware #119: [Bug]: Connection errors for Quest 3

walterleonardo, 2024-06-03:
### What happened?

I did install the last version 0.7.0-rc1 and work perfectly in IOS and Android mobiles phone. But in quest, are detected correctly in bluetooth but into the Haptic player don't... and don't work in games. 

I have a screen shot that I report the Version of the Bhaptic player and the version of Quest are 66. 
Is detected by the BHaptic, can see into the list, but can't connect and use it. 

Greetings.   

### Target Device

x16 haptic

### Hardware Configuration

ESP32-WROOM-32 

### Firmware Version

0.7.0 RC1

### What OS or environment are you seeing the problem on?

Meta Quest

### bHaptics Player Version

1.2.3 

### Relevant log output

```Shell
Quest 3 version 66.
```

### Serial output

```Shell
NO serial output
```

### Contact Details

walterleonardo@gmail.com

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

walterleonardo, 2024-06-03:
[image]
[image]

walterleonardo, 2024-06-03:
How you can see into the screenshot, the Hptic16x is detected by quest 3, and the Bhapticplayer can see the MacAddress. 
But Is like disconnected. From the APP BhapticPlayer. 
I test with Main, firmware and 0.7.0rc1

walterleonardo, 2024-06-03:
In android mobile and Ios mobiles, the Haptic player work perfectly .
I can't test in PC I only have MAC.

leon0399, 2024-06-04:
Thank you for reporting this bug! I'll try to replicate it!
In the meantime, have you tried connecting with different firmware versions? 0.6.* or any other?

walterleonardo, 2024-06-04:
Yes.. I already test the version 0.6.0 but doesn't work... 
I think that maybe the Oculus version... is the only that I change... from the last time that I test it. 

I already test your design 2 month ago... and worked perfectly. but Now and with the last version from Oculus Quest3 Doesn't work.

walterleonardo, 2024-06-16:
Did I recreate the hardware and installed the last firmware--- 0-7-0.rc1, but in this case, I create the version with Driver PCA9685. 
[bhaptics_tactsuit_x16_pca9685.zip](https://github.com/senseshift/senseshift-firmware/releases/download/v0.7.0-rc.1/bhaptics_tactsuit_x16_pca9685.zip)

Whit this hardware:
https://github.com/senseshift/senseshift-hardware/tree/main/Devices/X16%20Vest/ESP32%20+%20PCA9685

But the same result.  Is detected by Oculus Quest 3... like Bhactipcx16 bluetooth device, but inside the applications don't detect it.

walterleonardo, 2024-06-17:
Tested in windows, and work...  0.7.0rc1
The only problem appear in Oculust Quest 3 v66

closed 2024-07-02 by leon0399

walterleonardo, 2024-07-03:
now is working perfectly both with Haptic player in Oculus Quest 3 V67......hardware version x16 with and without PCA9685

Today at night I will test it with the game. Thx by the effort.
Note: Can you include a var FakeBatteryLevel???
To simulate a value... to see like battery are charged, in some games with Battery checker, will generate problems. 

debug
--- Terminal on /dev/cu.SLAB_USBtoUART | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[   734][I][BLEDevice.cpp:577] getAdvertising(): create advertising
[100469][I][senseshift.cpp:21] postEvent(): [application] Event dispatched at 100469: connected (0x3ffc805c)
[100650][I][BLEDevice.cpp:222] gapEventHandler(): ESP_GAP_BLE_SEC_REQ_EVT
[102714][I][BLEDevice.cpp:249] gapEventHandler(): key type = ESP_LE_KEY_LENC
[102722][I][BLEDevice.cpp:249] gapEventHandler(): key type = ESP_LE_KEY_PENC
[102730][I][BLEDevice.cpp:249] gapEventHandler(): key type = ESP_LE_KEY_LID
[102739][I][BLEDevice.cpp:249] gapEventHandler(): key type = ESP_LE_KEY_LCSRK
[102844][I][BLEDevice.cpp:249] gapEventHandler(): key type = ESP_LE_KEY_PID
[102852][I][BLEDevice.cpp:249] gapEventHandler(): key type = ESP_LE_KEY_PCSRK
[102940][I][BLEDevice.cpp:253] gapEventHandler(): ESP_GAP_BLE_AUTH_CMPL_EVT
[104675][I][senseshift.cpp:21] postEvent(): [application] Event dispatched at 104675: disconnected (0x3ffbd6f4)
[114959][I][senseshift.cpp:21] postEvent(): [application] Event dispatched at 114959: connected (0x3ffc822c)
[115286][I][BLEDevice.cpp:253] gapEventHandler(): ESP_GAP_BLE_AUTH_CMPL_EVT
[181853][I][senseshift.cpp:21] postEvent(): [application] Event dispatched at 181853: disconnected (0x3ffbd3e4)

[image]
[image]

leon0399, 2024-07-07:
@walterleonardo as I stated in https://github.com/senseshift/senseshift-firmware/issues/121#issuecomment-2182325241, currently, the battery uses a different configuration compared to schematics

I've just changed the default battery level to 100% in the code, so if you want to use it, you can flash the recent `master` in PlatformIO IDE, and it will always report full battery

commit b5efe58d (2024-07-02): fix(bHaptics): add GlobalConfig char

Fixes #119
commit 42021f83 (2024-07-02): fix(bHaptics): add GlobalConfig char

Fixes #119
PR #122 (2024-07-02): fix(bHaptics): add GlobalConfig char
Fixes #119
PR #123 (2024-07-02): fix(bHaptics): connect to Quest 3 on 0.7.0
Fixes #119